mithi/hexapod-robot-simulator #17: Hexapod doesn't check where the angles are within range as set by hexapod.MAX_ANGLES

mithi, 2020-04-05:
- It should

markkulube, 2020-06-18:
@mithi I would like to help - but could you be a little more clearer what it is that needs to be changed and where exactly.

Right now, I feel like I have to guess a little bit.....OR I just don't understand!

mithi, 2020-06-18:
Thanks for your interesting in making helping with this project!

What I meant was, to check it the `poses` are within range before updating the `VirtualHexapod`
 
https://github.com/mithi/hexapod-robot-simulator/blob/ccbdfdb91ece0b9b719f3733f884385cb0f6896e/hexapod/models.py#L119

You can make a helper function ie 
```
def poses_within_range(pose):
```

which can be called 
```
if (poses_within_range(pose):
    raise Exception(f"❗Some of the values in {poses} are not within range")
``` 

or something like that.

Hope that helps!

Don't forget to test your code to make sure you didn't break anything! Enjoy!

mithi, 2020-06-18:
> @mithi I would like to help - but could you be a little more clearer what it is that needs to be changed and where exactly.
> 
> Right now, I feel like I have to guess a little bit.....OR I just don't understand!

 Thanks for your interesting in making helping with this project! 

What I meant was, to check it the `poses` are within range before updating the `VirtualHexapod`
 
https://github.com/mithi/hexapod-robot-simulator/blob/ccbdfdb91ece0b9b719f3733f884385cb0f6896e/hexapod/models.py#L119

You can make a helper function ie 
```
def poses_within_range(pose):
```

which can be called 
```
if (poses_within_range(pose):
    raise Exception(f"❗Some of the values in {poses} are not within range")
``` 

or something like that.

Hope that helps!

Don't forget to test your code to make sure you didn't break anything! Enjoy!

markkulube, 2020-06-22:
@mithi i was having trouble logging into git for the past days, but seeing your note now.

thanks, for the update, i think i get it and can get to work.

markkulube, 2020-06-22:
@mithi a couple issues here:

1. self.legs[i].change_pose(pose["coxia"], pose["femur"], pose["tibia"]) is updating the dimensions of the robot NOT the leg pattern angles. (models.py - line 127)
2. the default dimensions are no assigned/declared in settings.py
3. def update(self, poses, assume_ground_targets=True): is NOT updating leg pattern angles as implied by this issue.

i am not asking to be spoon fed, but it looks like there is a bit of contradiction, if you can clarify when you get a moment.

thanks!

mithi, 2020-06-22:
Hi @markkulube ,

I'm sorry but I don't think you understand how the program works. Your first and third point is incorrect. 

First of all, I'm sorry that the naming convention is confusing you. I understand where the confusion is coming from. This is because `femur`, `coxia` and `tibia` names are referring to the lengths / dimensions of the hexapod in the dimensions dictionary. But `femur`, `coxia` and `tibia` in the poses and pattern poses dictionary are referring to the angles made by each pose. 

You will be able to infer this information by looking at the following files: 

The `coxia`, femur`, and `tibia`, here are referring to lengths of the hexapod
https://github.com/mithi/hexapod-robot-simulator/blob/ccbdfdb91ece0b9b719f3733f884385cb0f6896e/hexapod/const.py#L10-L17

The `coxia`, `femur`, and `tibia` is referring to angles here in degrees
https://github.com/mithi/hexapod-robot-simulator/blob/ccbdfdb91ece0b9b719f3733f884385cb0f6896e/hexapod/templates/pose_template.py#L1-L19

If you look at the pattern pose page

https://github.com/mithi/hexapod-robot-simulator/blob/ccbdfdb91ece0b9b719f3733f884385cb0f6896e/pages/page_patterns.py#L59-L61

You will see it invokes the `make_pose` function, which assigns the `alpha`, `beta` and gamma` to each leg respectively.

https://github.com/mithi/hexapod-robot-simulator/blob/ccbdfdb91ece0b9b719f3733f884385cb0f6896e/pages/helpers.py#L20-L30

Which means it will be
```python

pose = { 
     0: {"coxia": alpha, "femur": beta, "tibia": gamma, "name": "right-middle", "id": 0}, 
     1: {"coxia": alpha, "femur": beta, "tibia": gamma, "name": "right-front", "id": 1}, 
     2: {"coxia": alpha, "femur": beta, "tibia": gamma, "name": "left-front", "id": 2}, 
     3: {"coxia": alpha, "femur": beta, "tibia": gamma, "name": "left-middle", "id": 3}, 
     4: {"coxia": alpha, "femur": beta, "tibia": gamma, "name": "left-back", "id": 4}, 
     5: {"coxia": alpha, "femur": beta, "tibia": gamma, "name": "right-back", "id": 5}, 
 } 

```

If you look at the kinematics and pattern pages of the app, you will find the following lines

https://github.com/mithi/hexapod-robot-simulator/blob/ccbdfdb91ece0b9b719f3733f884385cb0f6896e/pages/page_kinematics.py#L46-L49

https://github.com/mithi/hexapod-robot-simulator/blob/ccbdfdb91ece0b9b719f3733f884385cb0f6896e/pages/page_patterns.py#L39-L42

Which is practically identical. 

It gives the pose, (which is a dictionary which contains the 18 poses, 3 angles for each of the 6 legs, 18 total) to the `update` method to update the points of the hexapod. 

You will also see it in this function which is a function that the inverse kinematics page calls. 

https://github.com/mithi/hexapod-robot-simulator/blob/ccbdfdb91ece0b9b719f3733f884385cb0f6896e/hexapod/ik_solver/recompute_hexapod.py#L25-L26

I'm sorry that all of this are confusing to you. This is my first-time creating an `first-timers-only` issue, I will be removing this the `first-timers-only` tag in this issue and closing this issue, as it seems that it's not as straightforward as I thought it would be given the non-trivial size of the project with has very little documentation.

markkulube, 2020-06-22:
@mithi THANK YOU for the very detailed explanation. I think with this i should be able to navigate better through the project.

That's reasonable, while this really a simple issue, it demands a quite a bit more than what is normally reserved for first-timers-only / good-first-issue tags.

Such issues are really meant to acquaint the first time with forking, cloning, and the PR sequence. For instance renaming coxia, femur, and tibia to distinguish between where said variables/keys refer angle and length/dimension. Obviously you would need to be explicit as to where the renaming takes place that is to the very line.

That being said, you've, put in a lot of effort in clarifying here, so I in want to see this issue through.

Thanks again!

mithi, 2020-06-24:
> @mithi THANK YOU for the very detailed explanation. I think with this i should be able to navigate better through the project..
> 
> That being said, you've, put in a lot of effort in clarifying here, so I in want to see this issue through.
> 
> Thanks again!

Enjoy! ❤️

markkulube, 2020-06-24:
@mithi Thanks. PS: I am also being stretched from my coop/internship. This still remains on my todo list, but please bear with me.

closed 2020-07-02 by mithi