mithi/hexapod-robot-simulator #68: Add docstrings for public packages etc

mithi, 2020-04-19:
# Public Packages
```
hexapod/
widgets/
pages/
tests/

```
# Some public methods
```
hexapod/ik_solver/ik_solver2.py
- init 

hexapod/linkage.py
- init
- str
- repr

hexapod/points.py
- init
- repr
- str
- eq

hexapod/models.py
- for VirtualHexapod
- for hexagon
```

See also: https://www.python.org/dev/peps/pep-0257/

mithi, 2020-05-24:
Particularly, fix the docstring of the following two modules so that the algorithm is easier to understand:

https://github.com/mithi/hexapod-robot-simulator/blob/master/hexapod/ground_contact_solver/ground_contact_solver.py

https://github.com/mithi/hexapod-robot-simulator/blob/master/hexapod/ground_contact_solver/ground_contact_solver2.py

SoniyaShan, 2020-07-14:
I want to claim this.

myiwt, 2020-09-03:
Hi, would you still like help with this?

memoisaaclo, 2020-10-06:
Hey I'm going to try my best to help you, but—disclaimer—I'm new and I'm not sure if it's going to be good.

mithi, 2020-10-06:
ok cool! feel free to ask me about anything!
and dont forget to read the contribution and commit style guide :)

memoisaaclo, 2020-10-20:
Sounds good!

pratheekcr, 2021-01-10:
Hi, I am new to open source and would like to contribute to this ticket. Is this still open?

mithi, 2021-01-10:
@pratheekcr 

hey yes, more and better test cases are always welcome!

feel free to ask me about anything! and dont forget to read the contribution and commit style guide!