mithi/hexapod-robot-simulator #75: ❗Wrong ground contacts behavior discovered

mithi, 2020-04-20:
[image]
[image]

mithi, 2020-04-20:
[image]

philippeitis, 2020-04-20:
This seems like it would choose the most parallel triangle, but it has a few problems (eg. it ignores more angled planes that are valid). I think that a more robust option might be to do the following instead:

Take all sets of any three points and their corresponding triangle where:
- The vector cast from the center of the hexapod to the center of the triangle is mostly perpendicular (+- some tolerance) - (the solutions formed by the current algorithm check for (0, 0, 0) being inside, but this doesn't check that the ray is perpendicular(ish) or close to the center))
- The triangle has a sufficiently large area (more stable)
- No points exist under the plane (so the sign of all their heights relative to the plane is correct)
   - This could possibly be used to search the problem space further (eg. take a point that is under the plane, and form a triangle with that point instead). 

Incidentally, this algorithm should allow solutions such as standing on two legs (eg. kneeling).
```
\  v---- body
  \  
--- <--- legs 1 / 2
```

This also allows vectorized numpy path to check the height of all points, which is nice.

mithi, 2020-04-21:
@philippeitis Thanks for your input, I'll take note of it and test it.

mithi, 2020-04-21:
# For documentation purposes, here's another WRONG algorithm 
```
Given:

find the lowest point given 18 points
(6 legs, 3 points each leg) that's the first point

now we have to find the second point from
the 15 candidate points (3 points each leg from 5 remaining legs)

draw a vector from the first point to each
of the 15 candidate points
get the angle between each vector and the
plane defined the hexapod's body

get the two lowest angles,
the two points corresponding to these two angles are the second and third point
https://www.quora.com/How-do-we-find-the-angle-between-a-vector-and-a-vector-plane-which-itself-is-made-by-two-vectors

assert that given the plane defined by these three points,
no other points are lower when the hexapod is reoriented
```

mithi, 2020-04-21:
# Algorithm I'm currently testing

```
We have 18 points total.

We have a total of 540 combinations
- get three legs out of six (20 combinations)
  - we have three possible points for each leg, that's 27 (3^3)
  -  27 * 20 is 540

For each combination:
    1. Check if stable if not, next
      - Check if the projection of the center of gravity to the plane
        defined by the three points lies inside the triangle, if not, next
      - https://math.stackexchange.com/questions/544946/determine-if-projection-of-3d-point-onto-plane-is-within-a-triangle
      - https://gamedev.stackexchange.com/questions/23743/whats-the-most-efficient-way-to-find-barycentric-coordinates
      - https://en.wikipedia.org/wiki/Barycentric_coordinate_system
    2. Get the height and normal of the height and normal of the triangle plane
        (We need this for the next part)
    3. For each of the three leg, check if the two other points on the leg is not a
        lower height, if condition if broken, next. (6 points total)
    4. For each of the three other legs, check all points (3 points of each leg)
        if so, next. (9 points total)
    5. If no condition is violated, then this is good, return this!
        (height, n_axis, 3 ground contacts)

```

mithi, 2020-04-21:
The current algorithm produces this results:

[image]
[image]
[image]

mithi, 2020-04-21:
Fixed with this commit 
https://github.com/mithi/hexapod-robot-simulator/commit/531fbb34d44246de3a0116def7e3d365de25b9f6

mithi, 2020-06-14:
[image]
[image]
[image]
[image]