mithi/hexapod-robot-simulator #80: Draw the projection of the center of gravity on the ground plane

mithi, 2020-04-21:


markkulube, 2020-07-08:
@mithi This has been open for a couple months now. 

Let me know when you think a "seasoned" contributor can take it on.....i would be interest!

mithi, 2020-07-08:
> @mithi This has been open for a couple months now.
> 
> Let me know when you think a "seasoned" contributor can take it on.....i would be interest!

Sure go ahead, if you have some time to tinker around ... I I currently don't have time to explain things in detail but by inspecting the code I'm sure you'll be able to figure things out on your own... 😄 

[here's the new guide to contributing.](https://github.com/mithi/hexapod/blob/master/CONTRIBUTING.md)

markkulube, 2020-07-08:
@mithi whaaatt? i like you detailed explanations, they get me thinking about clean code.

in all seriousness, thank you.

i only hope/expect  can chime in when, and however you can.

so no pressure.

markkulube, 2021-05-21:
Hi @mithi. Apologies for the very very very long delay. I am finally done with school, can get back to OSS contributions, and I can start working on this issue.

mithi, 2021-05-22:
@markkulube 

Hey! Long time no see 😄 This issue is open, so if you'd like to work on it, you're more than welcome 🙂

markkulube, 2021-07-01:
@mithi  a couple short questions:

1. What information is present in `dimensions` that I can use to initialize the point of projection for the  self.projection_point  attribute? The initial value is None then or is it a Point/Vector as per points.py?
` class VirtualHexapod: 
    def __init__(self, dimensions):
            self._store_attributes(dimensions)
            self._init_legs()
            self._init_local_frame()
            self.projection_point = None
` 
2. I understand calculation of the point is to be abstracted to it's own function e.g. compute_projection_point(---), but what are it's parameters? It would really help if there was a mathematical formula you had in my that could be implemented in the code!

mithi, 2021-07-03:
@markkulube 

`VirtualHexapod` has a property `self.body` which is a `Hexagon`

https://github.com/mithi/hexapod-robot-simulator/blob/531fbb34d44246de3a0116def7e3d365de25b9f6/hexapod/models.py#L61

Hexagon has a property `self.cog` this is the center of gravity which is of type point `Point`

The projection of the center of gravity is just a point where the z coordinate is zero. IE
```
cogProjection.x = cog.x
cogProjection.y = cog.y
cogProjection.z = 0
```

Thanks!