mithi/hexapod-robot-simulator #104: Add more and better tests and test cases

mithi, 2020-06-17:
Get started with testing python
https://realpython.com/python-testing/

Write Professional Unit Tests in Python
- https://code.tutsplus.com/tutorials/write-professional-unit-tests-in-python--cms-25835

Web Automation Tests with Selenium
- https://www.browserstack.com/guide/python-selenium-to-run-web-automation-test

Veluxer, 2020-06-23:
Hello =) i would be interested by helping as i know a bit of python and i used selenium for a personal project, did you code this under windows or linux? and would you mind telling me if there are specific step to take to help (or clone the repository)?
i thought your project would be interesting but i never contributed before >.< so i'm sorry if my question seem dumb ^^

mithi, 2020-06-23:
@Veluxer !

Thank you for your interest in contributing to this project!

I tested this code on macOS, but it is currently deployed on Heroku which uses lightweight Linux containers 
see also:
- https://www.heroku.com/dynos
- https://dash.plotly.com/deployment

To get started, you can do the following:

### 1. Clone this repo
```
git clone https://github.com/mithi/hexapod-robot-simulator.git
```

### 2. Inside the repo, install the requirements
```
pip3 install -r requirements.txt
```
If you have trouble you can checkout stackoverflow
https://stackoverflow.com/questions/41457612/how-to-use-requirements-txt-to-install-all-dependencies-in-a-python-project?rq=1

The other requirements are mostly for deployment, the only things you really need to install are the following:

```
Python 3.8.1
Plotly Dash 1.10.0
Plotly Dash Daq 0.4.0
Numpy 1.18.1
```

Which can easily be installed with
```
pip3 install dash==1.13.3
pip3 install dash_daq
pip3 install numpy
```
### 3. Check if it runs
```
python index.py
```
You should be able to see it running on your browser at `Running on http://127.0.0.1:8050/`

# Help creating tests

As of the moment, this project has poor test coverage (as you can see it's only about `62%` this means that a lot of blocks of codes are not executed when our current tests are run.

What kinds of tests to create is up to you but I suggest the best ones are the most basic ones, like making sure the page renders properly, and that the links, sliders and text boxes are working and stuff like that. 

It would also be nice if more test cases are added
- https://github.com/mithi/hexapod-robot-simulator/tree/master/tests

example test cases
- https://github.com/mithi/hexapod-robot-simulator/blob/master/tests/ik_cases/case1.py
- https://github.com/mithi/hexapod-robot-simulator/blob/master/tests/kinematics_cases/case1.py

Let me know if you have any more questions! Thanks!

Veluxer, 2020-06-24:
Thanks a lot!!!
That was quite clear =)
i will try and reach out if i have any question!

mithi, 2020-06-24:
> Thanks a lot!!!
> That was quite clear =)
> i will try and reach out if i have any question!

Enjoy! ❤️

GobindB, 2020-07-20:
Hey there, I'm looking for first issues to get involved with as I foray into the world of open-source :)  I wanted to see if there was still scope to get involved with solving this issue? Thank you for taking the time!

mithi, 2020-07-20:
> Hey there, I'm looking for first issues to get involved with as I foray into the world of open-source :) I wanted to see if there was still scope to get involved with solving this issue? Thank you for taking the time!

Yes! :)

Please read: https://github.com/mithi/hexapod-robot-simulator/issues/104#issuecomment-648408682

and you can check out the alerts that could be triggered in this file
https://github.com/mithi/hexapod-robot-simulator/blob/master/hexapod/ik_solver/helpers.py

Please let me know if you have any questions. Thanks!

RonakSharma1, 2020-08-13:
Hi! 
I am currently looking for opportunities to contribute to Github and saw that this project is a first-timer issue. Therefore, I wanted to know if this issue is still open to contribute. Thanks

mithi, 2020-08-13:
> Hi!
> I am currently looking for opportunities to contribute to Github and saw that this project is a first-timer issue. Therefore, I wanted to know if this issue is still open to contribute. Thanks

Yes, this is still open! Feel free to ask me anything if there is anything you need clarified! Thanks!

myiwt, 2020-09-03:
Hi there, would you still like some more help on this?

mithi, 2020-09-03:
> Hi there, would you still like some more help on this?

Yes, this is still open! Feel free to ask me anything if there is anything you need clarified! Thanks!

mithi, 2020-09-04:
Sure! Having more and better tests is always a good thing!

sgetty, 2020-12-04:
Hi, I am looking for a good opportunity to contribute to a GitHub Repository. Is this issue still open for contributions? Thanks.

mithi, 2020-12-05:
> Hi, I am looking for a good opportunity to contribute to a GitHub Repository. Is this issue still open for contributions? Thanks.

Yup! Still open!

Please checkout the [Contribution Guide](https://github.com/mithi/mithi/wiki/Contributing) and [commit style guide](https://github.com/mithi/mithi/wiki/Commit-style-guide)

Have fun!

pratheekcr, 2021-01-10:
HI @mithi is this ticket still open? I would love to contribute, please let me know.

mithi, 2021-01-10:
> HI @mithi is this ticket still open? I would love to contribute, please let me know.

Hi @pratheekcr,

More and better test cases are always welcome!

Don't forget to checkout the [Contribution Guide](https://github.com/mithi/mithi/wiki/Contributing) and [commit style guide](https://github.com/mithi/mithi/wiki/Commit-style-guide)

Have fun!